uzh-rpg/flightmare #121: How to timestamp ROS image messages

fletsch, 2021-04-26:
Hi. I have been running the ROS `flight_pilot` node from the together with publishing rgb, segmentation and depth image as described in [the tutorial](https://flightmare.readthedocs.io/en/latest/tutorials_general/camera.html#tut-camera) which worked fine so far. Now I need the timestamp of the image messages to be synchronized with the pose messages from gazebo (i.e. a rgb image retrieved at a certain pose should have the same timestamp as this pose). One way I have tried to achieve this is to retrieve and publish the images in the pose callback (code below) and assign them the current stamp of the pose. What I observe now is that the image seems not to belong to exactly this pose but rather to a previous one. As I don't fully understand the timing of the zmq messaging in getRender and handleOutput my two questions would be:
1. Do you think this behaviour is possible?
2. And if yes, do you have an idea on how to properly timestamp the image messages such that they match the pose they were retrieved at?

Thanks a lot!

```
void FlightPilot::poseCallback(const nav_msgs::Odometry::ConstPtr &msg) {
  quad_state_.x[QS::POSX] = (Scalar)msg->pose.pose.position.x;
  quad_state_.x[QS::POSY] = (Scalar)msg->pose.pose.position.y;
  quad_state_.x[QS::POSZ] = (Scalar)msg->pose.pose.position.z;
  quad_state_.x[QS::ATTW] = (Scalar)msg->pose.pose.orientation.w;
  quad_state_.x[QS::ATTX] = (Scalar)msg->pose.pose.orientation.x;
  quad_state_.x[QS::ATTY] = (Scalar)msg->pose.pose.orientation.y;
  quad_state_.x[QS::ATTZ] = (Scalar)msg->pose.pose.orientation.z;
  //
  quad_ptr_->setState(quad_state_);

  if (unity_render_ && unity_ready_) {
    unity_bridge_ptr_->getRender(0);
    unity_bridge_ptr_->handleOutput();

    if (quad_ptr_->getCollision()) {
      // collision happened
      ROS_INFO("COLLISION");
    }
  }

  cv::Mat img;
  ros::Time timestamp =  msg->header.stamp;

  rgb_camera_->getRGBImage(img);
  sensor_msgs::ImagePtr rgb_msg =
    cv_bridge::CvImage(std_msgs::Header(), "bgr8", img).toImageMsg();
  rgb_msg->header.stamp = timestamp;
  rgb_pub_.publish(rgb_msg);

  rgb_camera_->getDepthMap(img);
  sensor_msgs::ImagePtr depth_msg =
    cv_bridge::CvImage(std_msgs::Header(), "32FC1", img).toImageMsg();
  depth_msg->header.stamp = timestamp;
  depth_pub_.publish(depth_msg);

  rgb_camera_->getSegmentation(img);
  sensor_msgs::ImagePtr segmentation_msg =
    cv_bridge::CvImage(std_msgs::Header(), "bgr8", img).toImageMsg();
  segmentation_msg->header.stamp = timestamp;
  segmentation_pub_.publish(segmentation_msg);
}
```

yun-long, 2021-04-30:
Hi,

In an ideal case, what you are doing is right. 
however, there is a huge delay in these two steps, especially since you have enabled the rgb, depth, and segmentation. 
```
    unity_bridge_ptr_->getRender(0);
    unity_bridge_ptr_->handleOutput();
```
this function doesn't guarantee that you will receive the image that is corresponding to the current `nav_msgs::Odometry::ConstPtr &msg` .
what might happen is that the image you receive is from the last time step because the image rendering for the current state
is not ready. 

What you can do is 

First,
// define send_frame_id_ = 0 as a class member of  your cpp class

unity_bridge_ptr_->getRender(frame_id);
int receive_frame_id =  unity_bridge_ptr_->handleOutput();
if (send_frame_id == recieve_frame_id) {
// it means the received image is corresponding to the current send pose, otherwise it was the image from last update
send_frame_id += 1;
}

remember also return the frame_id in this function
https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/bridges/unity_bridge.cpp#L208

in stead of doing `return true`, but do ''return sub_msg.frame_id;'

hope it helps

fletsch, 2021-05-02:
Yes this resolved the issue. Thanks for the help!

closed 2021-05-02 by fletsch