uzh-rpg/flightmare #122: Simulator very slow

brunopinto900, 2021-04-28:
Hello,

When i launch flightmare with "roslaunch flightros rotors_gazebo.launch" everything works fine, but the quadrotor moves very slowly and the simulator lags a lot. I can run AirSim on Unreal4 without lagging. I really want to use use flightmare on my racing drone project.

Here is my specs:

CPU: Intel Core™ i7, 2.80 GHz, Quad core
GPU: NVIDIA® GeForce® GTX 1050 Ti 4GB
RAM: 16 GB.

Is there something i could do on my NVIDIA settings to increase the speed of the simulator?

yun-long, 2021-04-30:
Hi, did you change image size? 

you can modify the image size here
https://github.com/uzh-rpg/flightmare/blob/master/flightros/src/pilot/flight_pilot.cpp#L30-L31

also, make sure you didn't compile your code in Debug mode.

panyaoqiang, 2025-01-10:
> I am facing the same issue, could you please tell what configurations we should use so that it runs smoothly?

Have you solved this problem? I am also facing the same problem. I am running 'roslaunch flight ros racing. launch' and it works, but I have found that my Nvidia 3060 is not working.

Tendertale, 2025-04-13:
> > I am facing the same issue, could you please tell what configurations we should use so that it runs smoothly?我也面临着同样的问题，你能告诉我我们应该使用什么配置才能让它平稳运行吗？
> 
> Have you solved this problem? I am also facing the same problem. I am running 'roslaunch flight ros racing. launch' and it works, but I have found that my Nvidia 3060 is not working.你解决了这个问题吗？我也面临着同样的问题。我正在运行“rosstart飞行ros比赛。发射”，它可以工作，但我发现我的Nvidia 3060不工作。

I have the same problem. It seems we might have to wait for a WSLg update from Microsoft to resolve this.