uzh-rpg/flightmare #128: [ddc_challenge] unbalance number of num_rollouts_per_density

beniaminopozzan, 2021-04-30:
Hi,

in
https://github.com/uzh-rpg/flightmare/blob/5136d7239dd33a73158388216705666304734296/flightrl/rpg_baselines/evaluation/evaluation.py#L99
each density value is repeated `num_rollouts_per_density` times. However in the next line
https://github.com/uzh-rpg/flightmare/blob/5136d7239dd33a73158388216705666304734296/flightrl/rpg_baselines/evaluation/evaluation.py#L100
the array is modified such that the first density is repeated `num_rollouts_per_density-1` times while the last one is repeated `num_rollouts_per_density+1` times. I can't understand the reason, in particular considering that during the analysis of the performances all the code is written assuming that the number of rollouts per density is fixed and equal to `num_rollouts_per_density` for all densities.

antonilo, 2021-05-03:
Should be solved now, or is something missing?

beniaminopozzan, 2021-05-04:
In my opinion, yes, it is.
Well, actually there was nothing to solve in the end.

closed 2021-05-04 by beniaminopozzan