uzh-rpg/flightmare #130: Passing waypoints to environment for training

huiyulhy, 2021-05-04:
Hi, looking through the python wrapper, I noticed that there is no function to set the waypoint as a goal for training even though the waypoint is passed in obs[12:14]. Is there a function to pass the waypoints into the flight env for training (i.e. to be used in the rewards calculation)?

erenleicter, 2022-02-17:
> Hi, looking through the python wrapper, I noticed that there is no function to set the waypoint as a goal for training even though the waypoint is passed in obs[12:14]. Is there a function to pass the waypoints into the flight env for training (i.e. to be used in the rewards calculation)?

 Did you find a way to set a goal for training?

evan-palmer, 2022-02-19:
@erenleicter The `goal_state_`, which includes the goal position, used by the reward function is initialized in the `QuadrotorEnv` initialization list (you can find this [here](https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/envs/quadrotor_env/quadrotor_env.cpp#L16)):
```
goal_state_((Vector<quadenv::kNObs>() << 0.0, 0.0, 5.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0).finished()),
```
I'm not aware of a setter method that can be used to update the goal state. If you want an easier way to configure the `goal_state_` that doesn't require you to rebuild `flightlib` each time you modify the desired position, you could add a new "goal_position" parameter to the `quadrotor_env.yml` configuration file and update `QuadrotorEnv` to set the "goal_position" as the position tracked in `goal_state_`.

I hope this helps.