uzh-rpg/flightmare #131: Possible alignment issue

francofusco, 2021-05-05:
I was creating a simple simulation and stumbled upon an issue that I suspect being related to Eigen and memory alignment. I am unsure if the issue is just on my side (bad compiler flags or whatnot) or if it is due to something missing/wrong in `flightlib`.

What I did was to compile Flightmare and another catkin package that uses `flightlib` in one executable to add a single drone inside the render. When running this program, everything was usually fine. However, sometimes it crashed without giving details. I recompiled everything in `Debug` mode and Eigen assertions were sometimes fired. Stupidly enough, I have not copied the messages, but I *think* they were related to memory alignment. I tried to pinpoint the issue, and it seems to be in [quadrotor.cpp](https://github.com/uzh-rpg/flightmare/blob/17778ad0985daf9171ac0ef239dbe423ac4572f4/flightlib/src/objects/quadrotor.cpp#L233). The call to `QuadrotorDynamics::getAllocationMatrix()` was successful, and the assertion was fired while copying (or moving?) the returned matrix inside `B_allocation_`. I had tried adding the following in my `CMakeLists.txt`:

```
add_executable(my_flightmare_example src/my_flightmare_example.cpp)
target_include_directories(my_flightmare_example PRIVATE ${catkin_INCLUDE_DIRS})
target_link_libraries(my_flightmare_example ${catkin_LIBRARIES} zmqpp stdc++fs)
target_compile_features(my_flightmare_example PRIVATE cxx_std_17)
```

The `target_compile_features` line was added in the hope of fixing the problem (I think I read somewhere that enabling C++17 takes care of Eigen's memory alignment issues). This was not the case however. Something that seemed to work was to add the option `Eigen::DontAlign` when defining `Vector` and `Matrix` inside `common/types.hpp`. This seemed to do the job, and I did not experience any crash for a while - possibly supporting the theory that the issue is related with memory alignment in Eigen. I have tried to remove all my changes and see if I could get the wrong behavior once again (in order to better explain this issue) but I haven't experienced another crash yet. As I said, this was occurring only once in a while.

Of course, it is just possible that I had tampered with `flightlib` and that it was entirely my fault. In this case, I apologize and hope this was not a huge loss of time. If instead you have any idea of what might be causing this and how to solve it, then any help is appreciated. And if it turns out to be an actual issue in `flightlib`, I am happy I could help! If I manage to reproduce the crash, I will post an update to help investigating the problem.

Cheers!

slimeth, 2021-06-06:
@francofusco let us know when it occurs again. To be honest, I had also already similar problems, but last time it was fixed with a different compiler flag if I remember correctly.

francofusco, 2021-06-07:
Ok, I'll keep the issue open for a while and see if at any point it happens again. As I said, it might have been due to me changing things all over the place while playing with flightlib... 😅

fcvm, 2022-01-29:
@francofusco Thank you! For me, ~4 of 5 times calling `std::make_shared<flightlib::Quadrotor>()` crashed. After changing https://github.com/uzh-rpg/flightmare/blob/170676e4dbc2cdbc1fb7566ab0238800902e3669/flightlib/include/flightlib/common/types.hpp#L26 to `using Matrix = Eigen::Matrix<Scalar, rows, cols, Eigen::DontAlign>;` it works consistently.