uzh-rpg/flightmare #136: Drone doesn't move

EpicDuckPotato, 2021-05-28:
Hi,
I just installed flightmare via ROS, and when I run

`roslaunch flightros rotors_gazebo.launch`

as specified in https://flightmare.readthedocs.io/en/latest/getting_started/quick_start.html#quick-start, I see a drone in Gazebo, a drone in the flightmare renderer, and a GUI. However, if I click connect in the GUI, start the autopilot, and tell the drone to go to z = 1 (set z = 1 in the GUI and press "Go To Pose") the drone doesn't do anything. The estimated state stays the same as well. Thoughts on why this is happening? Thanks!

slimeth, 2021-06-06:
Do you still experience this bug? Could you give some details about your system setup?

EpicDuckPotato, 2021-06-07:
I'm running Ubuntu 18.04 and ROS Melodic. I still have the bug, though if I write my own drone control node using the unity bridge API, I'm able to control the drone as expected.

souravsanyal06, 2022-06-24:
Was any fix found for this ?