uzh-rpg/flightmare #137: Better documentation for segmentation

EpicDuckPotato, 2021-05-29:
Hi,
I'm interested in using the segmentation API. Essentially, I want to spawn my own objects and identify pixels corresponding to that object using the segmentation. However, I am finding it unclear how to determine which color in the segmented image corresponds to my object. AirSim provides nice documentation on how to do this (https://microsoft.github.io/AirSim/image_apis/), and I'm wondering whether you can do the same. Thanks!

slimeth, 2021-06-06:
On the unity side, we run the [Image Synthesis from Unity](https://bitbucket.org/Unity-Technologies/ml-imagesynthesis/wiki/Implementation). It uses a special replacement shader, but for more details, you will need to dive into the code and Unity references yourself for now.

EpicDuckPotato, 2021-06-07:
Yeah, this is what I ended up doing. Thanks, though.