uzh-rpg/flightmare #145: Stuck in splash screen when using with ROS - No Connect in Unity

JulianKu, 2021-09-01:
Hi,
I've encountered an issue when trying to use _flightmare_ with ROS. After following the instructions, I run `roslaunch flightros rotors_gazebo.launch`. Consequently, the `RPG Quadrotor Gui` (_rqt_ window), Gazebo, and the `RPG_Flightmare.x86_64` Unity binary open. However, I cannot get the drone to start nor are any messages published by _flightmare_.

1) The Unity binary stays in the splash screen (I can change scenes, go to _Scene Preview_ and _Scene Save Pointcloud_ modes and _Quit_ also works. However, clicking on _Connect_ simply has no effect at all).
2) In Gazebo, a drone model is spawned at the Origin and keeps staying there.
3) In the _rqt_ GUI window I am able to _Connect_ (meaning that after clicking the button, all fields are filled with values) but when I try to start the drone, I get an error message like this:
`[ERROR] [1630477836.476653587, 47.710000000]: [/hummingbird/autopilot] No state estimate available, will not start`.

I am working on **Ubuntu 20.04**, **ROS Noetic** and downloaded the **latest Flightmare Unity binary** (i.e. `v0.0.5`)

This is the console output from running `roslaunch flightros rotors_gazebo.launch`: [https://pastebin.com/xc2q6dXC](https://pastebin.com/xc2q6dXC)

Any ideas on how to get this to work are very welcome!

JulianKu, 2021-09-01:
Might be related to #136

souravsanyal06, 2022-06-23:
Hi! Were you able to resolve this issue ?

JulianKu, 2022-06-28:
Unfortunately, no