uzh-rpg/flightmare #150: Adding new sensors (such as radar)

JuanshuB2, 2021-11-16:
Hi,
I am looking for the best simulator to test sensor fusion algorithms. One of these sensor is the radar (FMCW) and I would like to know whether is possible to add a sensor in Flightmare. If so, is there any radar sensor already available?
Thanks

JinMohi, 2024-12-21:
Hi，I am meeting the same question, have you solved it?

JuanMorogit, 2024-12-21:
Hi, this kind of radar was not available as far as I searched it nor in Flightmare neither other drone simulator, so we had to simulate it based on the actual physical equations. You can find our work published in https://ieeexplore.ieee.org/document/10061372 to see if it suits you. There is no code avaiable, we made it in Python from the actual data from the simulator.