uzh-rpg/flightmare #151: Error occurred when using command "pip install ."

luffyzik, 2021-11-17:
I am having some trouble installing Flightmare for ubuntu18.04.
when I type the command:
`when I type the command:`
is giving an error:
`Building wheels for collected packages: flightgym
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/luffyzik/anaconda3/envs/ENVNAME/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-f3kft0sj/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-f3kft0sj/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-z9m_3y8n
       cwd: /tmp/pip-req-build-f3kft0sj/
  Complete output (116 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 7.5.0
  -- The CXX compiler identification is GNU 7.5.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- No sufficient Eigen version (3.3.4) found.
  -- Restoring to download Eigen sources.
  -- Getting Eigen...
  Cloning into 'eigen3'...
  Note: checking out '3.3.4'.
  
  You are in 'detached HEAD' state. You can look around, make experimental
  changes and commit them, and you can discard any commits you make in this
  state without impacting any branches by performing another checkout.
  
  If you want to create a new branch to retain commits you create, you may
  do so (now or later) by using -b with the checkout command again. Example:
  
    git checkout -b <new-branch-name>
  
  HEAD is now at 3dc3a0ea2 bump to 3.3.4
  -- Eigen downloaded!
  -- Eigen3 include dir:
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.9.0 dev1
  -- Found PythonInterp: /home/luffyzik/anaconda3/envs/ENVNAME/bin/python (found version "3.6.13")
  -- Found PythonLibs: /home/luffyzik/anaconda3/envs/ENVNAME/lib/libpython3.6m.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
  -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
  -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
  -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
  -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
  -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
  -- Found OpenCV: /usr (found version "3.2.0")
  -- Found OpenMP_C: -fopenmp (found version "4.5")
  -- Found OpenMP_CXX: -fopenmp (found version "4.5")
  -- Found OpenMP: TRUE (found version "4.5")
  INFOBuild time could be improved with ccache!
  INFO    sudo apt install ccache
[... 147 lines cut ...]
        cmd_obj.run()
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/command/install.py", line 545, in run
        self.run_command('build')
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-req-build-f3kft0sj/setup.py", line 59, in run
        self.build_extension(ext)
      File "/tmp/pip-req-build-f3kft0sj/setup.py", line 89, in build_extension
        cmake_args, cwd=self.build_temp, env=env)
      File "/home/luffyzik/anaconda3/envs/ENVNAME/lib/python3.6/subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-f3kft0sj', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-f3kft0sj/build/lib.linux-x86_64-3.6/', '-DPYTHON_EXECUTABLE=/home/luffyzik/anaconda3/envs/ENVNAME/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/luffyzik/anaconda3/envs/ENVNAME/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-f3kft0sj/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-f3kft0sj/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-cnwpmz_q/install-record.txt --single-version-externally-managed --compile --install-headers /home/luffyzik/anaconda3/envs/ENVNAME/include/python3.6m/flightgym Check the logs for full command output.
`

LIU-Xueming, 2021-11-19:
Hi, 
I also encountered the same problem, have you solved it?

LIU-Xueming, 2021-11-22:
I tried  with this issues #111 , and when I execute the statement "cmake ..", an error is reported that google-test cannot be found, so I modified the file flightmare/flightlib/cmake/gtest_download.cmake ：“GIT_TAG           master” -> "GIT_TAG           main", then run pip install . under /flightmare/flightlib. it works for me. after that run  pip install -e . under flightrl Instead of pip install .     refer to #85  or #93 。
fixed it for me.

luffyzik, 2021-11-24:
> 

非常感谢，我也解决了

PengqinWANG, 2021-12-22:
Useful solution! Thank you! That works for me.

monkeyQsUn, 2022-01-12:
> I tried with this issues #111 , and when I execute the statement "cmake ..", an error is reported that google-test cannot be found, so I modified the file flightmare/flightlib/cmake/gtest_download.cmake ：“GIT_TAG master” -> "GIT_TAG main", then run pip install . under /flightmare/flightlib. after that run pip install -e . under flightrl Instead of pip install . refer to #85 or #93 。 fixed it for me.

thanks mate! this works for me too.

rm-azou, 2022-03-01:
> I tried with this issues #111 , and when I execute the statement "cmake ..", an error is reported that google-test cannot be found, so I modified the file [...]

Thaks! It really helps me a lot, but when I run pip install . under /flightmare/flightlib, it also failed, I run pip install -e . install under /flightmare/flightlib, it works.

wuliang020, 2022-03-19:
> I tried with this issues #111 , and when I execute the statement "cmake ..", an error is reported that google-test cannot be found, so I modified the file [...]

Scanning dependencies of target gtest
[ 43%] Built target yaml-cpp
[ 45%] Building CXX object ../externals/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 78%] Built target flightlib
[ 80%] Built target flightgym
[ 82%] Linking CXX static library ../../../build/lib/libgtest.a
[ 82%] Built target gtest
Scanning dependencies of target gtest_main
[ 83%] Building CXX object ../externals/googletest-build/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 84%] Linking CXX static library ../../../build/lib/libgtest_main.a
[ 84%] Built target gtest_main
[ 86%] Linking CXX executable test_unity_bridge
[ 87%] Linking CXX executable test_lib
/usr/bin/ld: libflightlib.a(vec_env.cpp.o): in function flightlib::VecEnv<flightlib::QuadrotorEnv>::init()': vec_env.cpp:(.text._ZN9flightlib6VecEnvINS_12QuadrotorEnvEE4initEv[_ZN9flightlib6VecEnvINS_12QuadrotorEnvEE4initEv]+0x25a): undefined reference to omp_set_num_threads'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_lib.dir/build.make:242: test_lib] Error 1
make[1]: *** [CMakeFiles/Makefile2:266: CMakeFiles/test_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 89%] Built target test_unity_bridge
make: *** [Makefile:141: all] Error 2

when I tried with this issues https://github.com/uzh-rpg/flightmare/issues/111 , it is successful when i cmake .. , but when i make -j4 found this error.