uzh-rpg/flightmare #155: catkin build error on Ubuntu20.04 port_def.inc: no such file or directory

PengqinWANG, 2021-12-27:
**My system is Ubuntu20.04, Gazebo11, ROS Noetic.
When my protobuf version < 3.7 like libprotoc 3.6.1 or 3.0.0, the result of catkin build shows:**
Errors     << rotors_gazebo_plugins:make /home/rhine/flight_ws/logs/rotors_gazebo_plugins/build.make.026.log                                                                                              
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/CommandMotorSpeed.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/CommandMotorSpeed.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:335：CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/ConnectGazeboToRosTopic.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/ConnectGazeboToRosTopic.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:348：CMakeFiles/mav_msgs.dir/ConnectGazeboToRosTopic.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Actuators.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Actuators.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:322：CMakeFiles/mav_msgs.dir/Actuators.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Float32.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Float32.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/FluidPressure.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/FluidPressure.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:374：CMakeFiles/mav_msgs.dir/Float32.pb.cc.o] error1
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:387：CMakeFiles/mav_msgs.dir/FluidPressure.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/ConnectRosToGazeboTopic.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/ConnectRosToGazeboTopic.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:361：CMakeFiles/mav_msgs.dir/ConnectRosToGazeboTopic.pb.cc.o] error 1
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:413：CMakeFiles/mav_msgs.dir/Imu.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Header.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Header.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:400：CMakeFiles/mav_msgs.dir/Header.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/JointState.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/JointState.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:426：CMakeFiles/mav_msgs.dir/JointState.pb.cc.o] error 1
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Lidar.pb.cc:4:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Lidar.pb.h:10:10: fatal error: google/protobuf/port_def.inc: no such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mav_msgs.dir/build.make:439：CMakeFiles/mav_msgs.dir/Lidar.pb.cc.o] error 1
make[1]: *** [CMakeFiles/Makefile2:1109：CMakeFiles/mav_msgs.dir/all] error 2

**When my protobuf version > 3.7 like libprotoc 3.7.1 or 3.19.1, the result of catkin build shows:**
usr/include/gazebo-11/gazebo/msgs/quaternion.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
17 | #error This file was generated by an older version of protoc which is
| ^~~~~
/usr/include/gazebo-11/gazebo/msgs/quaternion.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
[... 178 lines cut ...]
from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.h:26,
from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:4:
/home/rhine/anaconda3/envs/tf/include/google/protobuf/wire_format_lite_inl.h: In instantiation of ‘static google::protobuf::uint8* google::protobuf::internal::WireFormatLite::InternalWriteMessageToArray(int, const MessageType&, google::protobuf::uint8*) [with MessageType = gazebo::msgs::Quaternion; google::protobuf::uint8 = unsigned char]’:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:642:52: required from here
/home/rhine/anaconda3/envs/tf/include/google/protobuf/wire_format_lite_inl.h:905:62: error: no matching function for call to ‘gazebo::msgs::Quaternion::InternalSerializeWithCachedSizesToArray(google::protobuf::uint8*&) const’
905 | return value.InternalSerializeWithCachedSizesToArray(target);
| ^
In file included from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.h:34,
from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:4:
/usr/include/gazebo-11/gazebo/msgs/quaternion.pb.h:143:30: note: candidate: ‘google::protobuf::uint8* gazebo::msgs::Quaternion::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’
143 | ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/quaternion.pb.h:143:30: note: candidate expects 2 arguments, 1 provided
In file included from /home/rhine/anaconda3/envs/tf/include/google/protobuf/map_type_handler.h:35,
from /home/rhine/anaconda3/envs/tf/include/google/protobuf/map.h:49,
from /home/rhine/anaconda3/envs/tf/include/google/protobuf/generated_message_table_driven.h:34,
from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.h:26,
from /home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:4:
/home/rhine/anaconda3/envs/tf/include/google/protobuf/wire_format_lite_inl.h: In instantiation of ‘static google::protobuf::uint8* google::protobuf::internal::WireFormatLite::InternalWriteMessageToArray(int, const MessageType&, google::protobuf::uint8*) [with MessageType = gazebo::msgs::Vector3d; google::protobuf::uint8 = unsigned char]’:
/home/rhine/flight_ws/build/rotors_gazebo_plugins/Imu.pb.cc:662:57: required from here
/home/rhine/anaconda3/envs/tf/include/google/protobuf/wire_format_lite_inl.h:905:62: error: no matching function for call to ‘gazebo::msgs::Vector3d::InternalSerializeWithCachedSizesToArray(google::protobuf::uint8*&) const’
905 | return value.InternalSerializeWithCachedSizesToArray(target);

I would appreciate it if I could receive your reply.
Thank you!

mfirlefyn, 2022-09-16:
Hi! I have run into the same issue. Did you find any solution for this issue? Thank you in advance. It is weird that nobody has commented on this issue yet.

PengqinWANG, 2022-09-16:
> Hi! I have run into the same issue. Did you find any solution for this issue? Thank you in advance. It is weird that nobody has commented on this issue yet.

Hi! I recommend you to use flightmare on Ubuntu18.04 since when I tried 18.04 everything worked well. And actually we have not found any solutions of this issue on Ubuntu20.04.

mfirlefyn, 2022-09-16:
Thanks for your recommendation. There was a conflict between protoc 3.6.4 and 3.0.0 after installing via apt and implementing the fix from the wiki. I tried with only the 3.0.0 installed and having the same version returned from usr/bin and usr/bin/local, but no luck with that either. I will try Ubuntu 18.04. It is a shame that it does not work on 20.04 though.

JiahengZeng, 2022-10-19:
Same problem!

mj2g20, 2022-10-19:
I have same issue in Ubuntu 18.04

mfirlefyn, 2022-10-20:
I actually found a workaround that may be able to help you out. The problem is that Ubuntu 20.04 inherently works with Gazebo 11, which is built with protoc 3.6.1. Protoc 3.0.0 is only used for Gazebo 9 on Ubuntu 18.04. 
If you would like to keep using 20.04, which I am still doing, you have 2 options imo:

JiahengZeng, 2022-10-21:
I really appreciate your sincere answer. Actually,  I'm trying to configure [this](https://github.com/uzh-rpg/agile_autonomy) project. However, it is explained here that Ubuntu 20.04 and ROS noetic are used(So I think gazebo11 is used defaulty). Such an issue also appear when I try to use "catkin_build". Do you have some good idea? Please let me know.
-------------------------------------------------------------
| #include <google/protobuf/port_def.inc>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

mfirlefyn, 2022-10-21:
You could check out this [comment](https://github.com/BVLC/caffe/issues/6527#issuecomment-429661350) to check whether there are no protobuf version discrepancies on your system.
Also, maybe you could try `catkin_make` instead of `catkin build`, sometimes it gives some more error information.

JiahengZeng, 2022-10-22:
thanks bro! Your answer solves my problem~

wtyngh, 2023-06-22:
> I actually found a workaround that may be able to help you out. The problem is that Ubuntu 20.04 inherently works with Gazebo 11, which is built with protoc 3.6.1. Protoc 3.0.0 is only used for Gazebo 9 on Ubuntu 18.04. If you would like to keep using 20.04, which I am still doing, you have 2 options imo:
> 
> 1. Use a docker image for Ubuntu 18.04 and follow the guide to install Flightmare, then use VNC viewer to use the graphical interfaces and actually use Flightmare
> 2. Use Ubuntu 20.04 without Gazebo 11 as only the gazebo_rotors simulator actually uses Gazebo 11, you can still use all the other launch scripts in flightros (e.g. roslaunch flightros camera.launch)
> 
> To facilitate the second option, you would have to remove all instances of protoc before attempting to build since you only want the inherent protoc 3.6.1. Let me know if you would like some help with that.

Hi! I believe I encountered the same issue with the protobuf version error. My system is Ubuntu20.04, Gazebo11, ROS Noetic.

I am wondering if you could provide more details about facilitating the second option without reinstalling Ubuntu 18.04. Thanks a lot!

mfirlefyn, 2023-06-22:
Hi
You should check my answer on Oct 21 2022. Use the link to check whether protoc/protobuf is installed on your system.

The easiest way to install flightmare is follow the install guide and skip the step where it says you need protoc 3.0.0. The protoc version that gets installed automatically works fine, you may need to install additional ros-noetic-**** packages whenever you get a compiler error pointing to the missing package.

whyscience, 2024-11-18:
Check your protoc --version, if it is not 3.6.1, rm it, or sudo apt-get install libprotobuf-dev protobuf-compiler --reinstall,
if this still not work, reinstall from src,

wget https://github.com/protocolbuffers/protobuf/releases/download/v3.6.1/protobuf-all-3.6.1.tar.gz \
    && tar xzf protobuf-all-3.6.1.tar.gz \
    && cd protobuf-3.6.1 \
    && ./autogen.sh \
    && ./configure \
    && make -j${proc_n} && sudo make install
then rm -rf Firmware/build, build again, it should work.