uzh-rpg/flightmare #158: ModuleNotFoundError: No module named 'flightgym' when running the python example

charlo1998, 2022-01-13:
Hi, after following the ros install steps and trying to run the example with pip, i get this error:

**(env) user:~/catkin_ws/src/flightmare/flightrl/examples$ python3 run_drone_control.py --train 0 --render 1**
WARNING:tensorflow:
The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
  * https://github.com/tensorflow/io (for I/O related ops)
If you depend on functionality not listed there, please file an issue.

Traceback (most recent call last):
  File "run_drone_control.py", line 21, in <module>
    from flightgym import QuadrotorEnv_v1
ModuleNotFoundError: No module named 'flightgym'

i had similar issues with modules like baselines and managed to fix those, following the other posts, but i cant fix this one.
thanks!

charlo1998, 2022-03-16:
I think i solved it by following #93.

samaila53, 2023-08-02:
Hi，I have the same error with you, do you solve it?

samaila53, 2023-08-02:
Hi, after following the flightmare install steps , i get this error:

Building wheels for collected packages: flightgym
Building wheel for flightgym (setup.py) ... error
ERROR: Command errored out with exit status 1:
command: /home/sam/anaconda3/envs/quad/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/sam/Bureau/flightmare/flightlib/setup.py'"'"'; file='"'"'/home/sam/Bureau/flightmare/flightlib/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(file) if os.path.exists(file) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-m2jlvmei
cwd: /home/sam/Bureau/flightmare/flightlib/
Complete output (106 lines):
running bdist_wheel
running build
running build_ext
Removing some cache file: /home/sam/Bureau/flightmare/flightlib/build/temp.linux-x86_64-3.6
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ====================== !Flightmare! ======================
-- ======> Setup Dependencies
-- Using system provided Eigen.
/usr/include/eigen3
-- Eigen3 include dir: /usr/include/eigen3
-- Getting Pybind11...
Cloning into 'pybind11-src'...
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
error: 3262 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'pybind11-src'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 1511 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'pybind11-src'...
Already on 'master'
-- Pybind11 downloaded!
-- pybind11 v2.12.0 dev1
-- Found PythonInterp: /home/sam/anaconda3/envs/quad/bin/python (found suitable version "3.6.13", minimum required is "3.6")
-- Found PythonLibs: /home/sam/anaconda3/envs/quad/lib/libpython3.6m.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Getting yaml-cpp...
-- Yaml downloaded!
-- Found OpenCV: /usr (found version "4.5.4")
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
INFOBuild time could be improved with ccache!
INFO sudo apt install ccache
-- ======> Setup Compilation
-- Enabling fast optimization flags!
-- The activated CXX RELEASE configuration is:
-DVERSION_INFO="0.0.1" -fPIC -Wall -DNDEBUG -fopenmp -Ofast -march=native
-- The activated CXX DEBUG configuration is:
-DVERSION_INFO="0.0.1" -fPIC -Wall -g -fopenmp
-- ======> Setup Build
-- Getting gtests...
Cloning into 'googletest-src'...
fatal: invalid reference: master
CMake Error at googletest-download/googletest-prefix/tmp/googletest-gitclone.cmake:40 (message):
Failed to checkout tag: 'master'
[... 20 lines cut ...]
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/anaconda3/envs/quad/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
self.run_command('build')
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/cmd.py", line 313, in run_command
self.distribution.run_command(command)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/command/build.py", line 135, in run
self.run_command(cmd_name)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/cmd.py", line 313, in run_command
self.distribution.run_command(command)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/Bureau/flightmare/flightlib/setup.py", line 59, in run
self.build_extension(ext)
File "/home/sam/Bureau/flightmare/flightlib/setup.py", line 89, in build_extension
cmake_args, cwd=self.build_temp, env=env)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/subprocess.py", line 311, in check_call
raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/home/sam/Bureau/flightmare/flightlib', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/sam/Bureau/flightmare/flightlib/build/lib.linux-x86_64-3.6/', '-DPYTHON_EXECUTABLE=/home/sam/anaconda3/envs/quad/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.

ERROR: Failed building wheel for flightgym

Ho do solve it?

BetterLL, 2023-08-03:
I'm so sorry, I've forgotten how to solve it. You can search other maybe have the same problem.

samaila53, 2023-08-04:
Traceback (most recent call last):
File "run_drone_control.py", line 21, in
from flightgym import QuadrotorEnv_v1
ModuleNotFoundError: No module named 'flightgym'

can someone help me please

RuiqiZhang99, 2024-04-05:
> Traceback (most recent call last): File "run_drone_control.py", line 21, in from flightgym import QuadrotorEnv_v1 ModuleNotFoundError: No module named 'flightgym'
> 
> can someone help me please

Hi, have you solved it now? I meet the same issue and can not find anything helpful. Thanks.