uzh-rpg/flightmare #159: failed to build flightgym when pip install .

BetterLL, 2022-03-17:
Hi everyone, when I use the command  pip install .  it raise errors like follow:

Building wheels for collected packages: flightgym
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/lxj/anaconda3/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-x9eryj05
       cwd: /tmp/pip-req-build-30pan_pf/
  Complete output (36 lines):
  running bdist_wheel
  running build
  running build_ext
  CMake Error: The current CMakeCache.txt directory /tmp/pip-req-build-30pan_pf/build/temp.linux-x86_64-3.8/CMakeCache.txt is different than the directory /home/lxj/catkin_ws/src/flightmare/flightlib/build/temp.linux-x86_64-3.8 where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
  CMake Error: The source "/tmp/pip-req-build-30pan_pf/CMakeLists.txt" does not match the source "/home/lxj/catkin_ws/src/flightmare/flightlib/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-req-build-30pan_pf/setup.py", line 94, in <module>
      setup(
    File "/home/lxj/anaconda3/lib/python3.8/site-packages/setuptools/__init__.py", line 165, in setup
      return distutils.core.setup(**attrs)
    File "/home/lxj/anaconda3/lib/python3.8/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/lxj/anaconda3/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 223, in run
      self.run_command('build')
    File "/home/lxj/anaconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/lxj/anaconda3/lib/python3.8/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/home/lxj/anaconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-req-build-30pan_pf/setup.py", line 59, in run
      self.build_extension(ext)
    File "/tmp/pip-req-build-30pan_pf/setup.py", line 88, in build_extension
      subprocess.check_call(['cmake', ext.sourcedir] +
    File "/home/lxj/anaconda3/lib/python3.8/subprocess.py", line 364, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-30pan_pf', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-30pan_pf/build/lib.linux-x86_64-3.8/', '-DPYTHON_EXECUTABLE=/home/lxj/anaconda3/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for flightgym
  Running setup.py clean for flightgym
Failed to build flightgym
Installing collected packages: flightgym
  Attempting uninstall: flightgym
    Found existing installation: flightgym 0.0.1
    Uninstalling flightgym-0.0.1:
      Successfully uninstalled flightgym-0.0.1
    Running setup.py install for flightgym ... error
    ERROR: Command errored out with exit status 1:
     command: /home/lxj/anaconda3/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-9b0nsd51/install-record.txt --single-version-externally-managed --compile --install-headers /home/lxj/anaconda3/include/python3.8/flightgym
         cwd: /tmp/pip-req-build-30pan_pf/
    Complete output (75 lines):
    running install
    running build
    running build_ext
    -- The C compiler identification is GNU 9.4.0
    -- The CXX compiler identification is GNU 9.4.0
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working C compiler: /usr/bin/cc - skipped
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
[... 47 lines cut ...]
        self.distribution.run_command(command)
      File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/lxj/anaconda3/lib/python3.8/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/home/lxj/anaconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/lxj/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-req-build-30pan_pf/setup.py", line 59, in run
        self.build_extension(ext)
      File "/tmp/pip-req-build-30pan_pf/setup.py", line 88, in build_extension
        subprocess.check_call(['cmake', ext.sourcedir] +
      File "/home/lxj/anaconda3/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-30pan_pf', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-30pan_pf/build/lib.linux-x86_64-3.8/', '-DPYTHON_EXECUTABLE=/home/lxj/anaconda3/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
  Rolling back uninstall of flightgym
  Moving to /home/lxj/anaconda3/lib/python3.8/site-packages/flightgym-0.0.1.dist-info/
   from /home/lxj/anaconda3/lib/python3.8/site-packages/~lightgym-0.0.1.dist-info
  Moving to /home/lxj/anaconda3/lib/python3.8/site-packages/flightgym.cpython-38-x86_64-linux-gnu.so
   from /tmp/pip-uninstall-msphy32j/flightgym.cpython-38-x86_64-linux-gnu.so
ERROR: Command errored out with exit status 1: /home/lxj/anaconda3/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-30pan_pf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-9b0nsd51/install-record.txt --single-version-externally-managed --compile --install-headers /home/lxj/anaconda3/include/python3.8/flightgym Check the logs for full command output.

I've tried the #151 but it doesn't work. How to solve it?

shuaikangma, 2022-03-19:
I guess it may caused by plasethe network setting, you didn't git clone the "pybind11-src" successfully. you can download the code and unzip it into correct place, and run `pip install .` again.

BetterLL, 2022-03-20:
> I guess it may caused by plasethe network setting, you didn't git clone the "pybind11-src" successfully. you can download the code and unzip it into correct place, and run `pip install .` again.

Thank you very much I've solved the problem.

Wangzai-hub, 2023-05-23:
Hello, I have encountered a similar issue to yours. How can I determine where to place the downloaded files