uzh-rpg/flightmare #190: issue with installing with pip

ZhongmouLi, 2023-07-06:
Hi Guys,

I am following the guide to install with pip https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip.

However, error happens when I run
`cd flightmare/flightlib
# it first compile the flightlib and then install it as a python package.
pip install .`

which is

`Building wheels for collected packages: flightgym, gym, opencv-python
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/robot_li/anaconda3/envs/flightmare/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/robot_li/Desktop/flightmare/flightlib/setup.py'"'"'; __file__='"'"'/home/robot_li/Desktop/flightmare/flightlib/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-jidefugx
       cwd: /home/robot_li/Desktop/flightmare/flightlib/
  Complete output (91 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 9.4.0
  -- The CXX compiler identification is GNU 9.4.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- Using system provided Eigen.
  /usr/local/include/eigen3
  -- Eigen3 include dir: /usr/local/include/eigen3
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.11.0 dev1
  -- Found PythonInterp: /home/robot_li/anaconda3/envs/flightmare/bin/python (found suitable version "3.6.13", minimum required is "3.6")
  -- Found PythonLibs: /home/robot_li/anaconda3/envs/flightmare/lib/libpython3.6m.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  -- Found OpenCV: /usr (found version "4.2.0")
  -- Found OpenMP_C: -fopenmp (found version "4.5")
  -- Found OpenMP_CXX: -fopenmp (found version "4.5")
  -- Found OpenMP: TRUE (found version "4.5")
  -- ======> Setup Compilation
 -- Enabling fast optimization flags!
  -- The activated CXX RELEASE configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -march=native
  -- The activated CXX DEBUG configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -g -fopenmp
  -- ======>  Setup Build
  -- Getting gtests...
  Cloning into 'googletest-src'...
  fatal: invalid reference: master
  CMake Error at googletest-download/googletest-prefix/tmp/googletest-gitclone.cmake:49 (message):
    Failed to checkout tag: 'master'
  
  
  make[2]: *** [CMakeFiles/googletest.dir/build.make:98: googletest-prefix/src/googletest-stamp/googletest-download] Error 1
  make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/googletest.dir/all] Error 2
  make: *** [Makefile:91: all] Error 2
  CMake Error at cmake/gtest.cmake:17 (message):
    Build step for googletest failed: 2
  Call Stack (most recent call first):
    CMakeLists.txt:192 (include)
-- Configuring incomplete, errors occurred!
  See also "/home/robot_li/Desktop/flightmare/flightlib/build/temp.linux-x86_64-3.6/CMakeFiles/CMakeOutput.log".
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/home/robot_li/Desktop/flightmare/flightlib/setup.py", line 106, in <module>
      zip_safe=False,
    File "/home/robot_li/.local/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/robot_li/.local/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/robot_li/Desktop/flightmare/flightlib/setup.py", line 59, in run
      self.build_extension(ext)
    File "/home/robot_li/Desktop/flightmare/flightlib/setup.py", line 89, in build_extension
      cmake_args, cwd=self.build_temp, env=env)
    File "/home/robot_li/anaconda3/envs/flightmare/lib/python3.6/subprocess.py", line 311, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '/home/robot_li/Desktop/flightmare/flightlib', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/robot_li/Desktop/flightmare/flightlib/build/lib.linux-x86_64-3.6/', '-DPYTHON_EXECUTABLE=/home/robot_li/anaconda3/envs/flightmare/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for flightgym
`

santurini, 2023-07-15:
You have to modify _flightmare/flightlib/cmake/gtest_download.cmake_ as follows:
```
cmake_minimum_required(VERSION 3.0.0)

project(googletest-download NONE)

include(ExternalProject)
ExternalProject_Add(googletest
  GIT_REPOSITORY    https://github.com/google/googletest.git
  GIT_TAG           main
  SOURCE_DIR        "${PROJECT_SOURCE_DIR}/externals/googletest-src"
  BINARY_DIR        "${PROJECT_SOURCE_DIR}/externals/googletest-build"
  CONFIGURE_COMMAND ""
  BUILD_COMMAND     ""
  INSTALL_COMMAND   ""
  TEST_COMMAND      ""
  UPDATE_DISCONNECTED ON
)
```

This because git has replaced the tag 'master' with 'main'.

savvas-ap, 2023-11-07:
Hi, after replacing master with main I get the following error:

In file included from ...../flightmare/flightlib/src/bridges/unity_bridge.cpp:1:
  ...../flightmare/flightlib/include/flightlib/bridges/unity_bridge.hpp:19:10: fatal error: zmqpp/zmqpp.hpp: No such file or directory
     19 | #include <zmqpp/zmqpp.hpp>
        |          ^~~~~~~~~~~~~~~~~
  compilation terminated.
  
  Any ideas on that?

santurini, 2023-11-08:
I had the same error, you have to compile zmqpp on your machine but this will consist in compiling also other packages. In the end it will still give you some error so I personally gave up. Good luck and let me know if you are successfull!

mrunaljsarvaiya, 2023-11-21:
Running into the same proble, @savvas-ap did you happen to find a solution?

savvas-ap, 2023-11-21:
@mrunaljsarvaiya unfortunately not. I tried installing flightmare on Ubuntu 22.04 that zmqpp is no longer supported. I believe that on Ubuntu 20.04 things might were going to be a bit more straightforward, however, I have not tried that yet.

savvas-ap, 2023-11-28:
@mrunaljsarvaiya you may want to take a look here:
https://github.com/uzh-rpg/flightmare/issues/1

mrunaljsarvaiya, 2023-11-28:
> @mrunaljsarvaiya you may want to take a look here: #1

Thank you! I actually managed to get it to work last week. In addition the fixes mentioned in #1, I had to switch to the dev/version_22 branch