uzh-rpg/flightmare #194: catkin bulid Error

zyy19971234, 2023-10-17:
Failed     << rotors_gazebo_plugins:make                    [ Exited with code 2 ]                              
Failed    <<< rotors_gazebo_plugins                         [ 12.2 seconds ]                                    
Abandoned <<< rotors_gazebo                                 [ Unrelated job failed ]                            
Abandoned <<< rotors_simulator                              [ Unrelated job failed ]                            
Abandoned <<< rpg_quadrotor_integration_test                [ Unrelated job failed ]                            
________________________________________________________________________________________________________________
Warnings   << flightros:make /home/yuanyuan/1-Code/catkin_ws/logs/flightros/build.make.001.log                  
/usr/bin/ld: warning: libopencv_core.so.4.5, needed by /home/yuanyuan/1-Code/catkin_ws/devel/lib/libflightlib.so, may conflict with libopencv_core.so.4.2
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5
/usr/bin/ld: warning: libopencv_features2d.so.4.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0, may conflict with libopencv_features2d.so.4.5
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5
/usr/bin/ld: warning: libopencv_features2d.so.4.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0, may conflict with libopencv_features2d.so.4.5
cd /home/yuanyuan/1-Code/catkin_ws/build/flightros; catkin build --get-env flightros | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

................................................................................................................
Finished  <<< flightros                                     [ 8.6 seconds ]                                     
[build] Summary: 31 of 35 packages succeeded.                                                                   
[build]   Ignored:   None.                                                                                      
[build]   Warnings:  32 packages succeeded with warnings.                                                       
[build]   Abandoned: 3 packages were abandoned.                                                                 
[build]   Failed:    1 packages failed.                                                                         
[build] Runtime: 27.5 seconds total.             

How to solve each problem？？？？？？？？【ubuntu20 gazebo11】

jhzhubit, 2023-12-12:
The same problem

AnshShah3009, 2024-02-17:
DId you resolve this? @ZyyFLY @jhzhubit

saadxsaleem, 2024-11-17:
How to solve this problem ? @jhzhubit