uzh-rpg/flightmare #201: Bodyrates and cumulative thrust as input for vec_env.step

somsomers, 2024-02-07:
Hello.

Could you please clarify how to use Flightmare with bodyrates+cumulative thrust instead of single motos thrusts?
I've created cmd_.omega matrix, disabled cmd_.thusts initialization so it runs runFlightCtl
`const Vector<4> motor_thrusts_des = cmd_.isSingleRotorThrusts() ? cmd_.thrusts : runFlightCtl(sim_dt, state_.w, cmd_);`

But the drone goes crazy even if I pass stable values each step, equal bodyrates and constant cumulative thrust(value range is like the one from "control_command" ROS topic, for example 0.01, 0.01, 0.01 and 9.1).

What am I doing wrong?

JzKosann, 2025-03-08:
Hello.
Are u still bothered by this question? 
u can open the flightrl run the example. By printing the thrust of vehicle, u can get the appropriate data.
Here is my question, would u help me?
I wonder how to control the vehicle?
I do
` // quad_ptr->setState(quad_state);
  Scalar t = 1.0;
  Vector<4> thrusts{500, 500, 500, 500}; 
  Command cmd(t,thrusts);
      if (cmd.valid()) {
        std::cout << "Command is valid." << std::endl;
    }

    if (cmd.isSingleRotorThrusts()) {
        std::cout << "Using single rotor thrusts mode." << std::endl;
        std::cout << "Thrusts: " << cmd.thrusts.transpose() << std::endl;
    }
  quad_ptr->setCommand(cmd);
`
but my vehicle dont response,

JzKosann, 2025-03-08:
> Hello.
>
> Could you please clarify how to use Flightmare with bodyrates+cumulative thrust instead of single motos thrusts? I've created cmd_.omega matrix, disabled cmd_.thusts initialization so it [...]
Hello.
Are u still bothered by this question?
u can open the flightrl run the example. By printing the thrust of vehicle, u can get the appropriate data.
Here is my question, would u help me?
I wonder how to control the vehicle?
I do
` // quad_ptr->setState(quad_state);
Scalar t = 1.0;
Vector<4> thrusts{500, 500, 500, 500};
Command cmd(t,thrusts);
if (cmd.valid()) {
std::cout << "Command is valid." << std::endl;
}

if (cmd.isSingleRotorThrusts()) {
    std::cout << "Using single rotor thrusts mode." << std::endl;
    std::cout << "Thrusts: " << cmd.thrusts.transpose() << std::endl;
}
quad_ptr->setCommand(cmd);
`
but my vehicle dont response,