uzh-rpg/flightmare #211: flightmare installation issues

abdul-mannan-khan, 2024-08-07:
Hello, Thank you so much for sharing this amazing work. Really grateful for this. I tried to install it following the installation guides give [here](https://flightmare.readthedocs.io/en/latest/getting_started/quick_start.html) and got following error:

```
Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/mannan/anaconda3/envs/flightmare/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-rd2n8tkg/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-rd2n8tkg/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_8w9qvlu
       cwd: /tmp/pip-req-build-rd2n8tkg/
  Complete output (73 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 9.4.0
  -- The CXX compiler identification is GNU 9.4.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- Using system provided Eigen.
  /usr/include/eigen3
  -- Eigen3 include dir: /usr/include/eigen3
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.14.0 dev1
  CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
    Could NOT find PythonInterp: Found unsuitable version "3.6.13", but
    required is at least "3.8" (found
    /home/mannan/anaconda3/envs/flightmare/bin/python)
  Call Stack (most recent call first):
    /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:391 (_FPHSA_FAILURE_MESSAGE)
    /usr/share/cmake-3.16/Modules/FindPythonInterp.cmake:169 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
    externals/pybind11-src/tools/FindPythonLibsNew.cmake:114 (find_package)
    externals/pybind11-src/tools/pybind11Tools.cmake:50 (find_package)
    externals/pybind11-src/tools/pybind11Common.cmake:202 (include)
    externals/pybind11-src/CMakeLists.txt:249 (include)
  
  
  -- Configuring incomplete, errors occurred!
  See also "/tmp/pip-req-build-rd2n8tkg/build/temp.linux-x86_64-3.6/CMakeFiles/CMakeOutput.log".
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-req-build-rd2n8tkg/setup.py", line 106, in <module>
      zip_safe=False,
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-req-build-rd2n8tkg/setup.py", line 59, in run
      self.build_extension(ext)
    File "/tmp/pip-req-build-rd2n8tkg/setup.py", line 89, in build_extension
      cmake_args, cwd=self.build_temp, env=env)
    File "/home/mannan/anaconda3/envs/flightmare/lib/python3.6/subprocess.py", line 311, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-rd2n8tkg', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-rd2n8tkg/build/lib.linux-x86_64-3.6/', '-DPYTHON_EXECUTABLE=/home/mannan/anaconda3/envs/flightmare/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for flightgym
```
Can you please take a look and update installation guide or repository? Thank you so much for your time.

lakshmikanth-nageswar, 2024-08-10:
Hi, I am facing similar issue and am unable to resolve it yet. Could you please let me know if you find a solution.

Thanks!

abdul-mannan-khan, 2024-08-10:
No. David Scaramuzza team needs to take a look at it as installation is very old.

purewoo1, 2024-09-15:
Hi, I am facing similar issue and am unable to resolve it yet. Could you please let me know if you find a solution.
Thanks!！！

Jinwei-Liu, 2024-10-15:
I encountered the same issues, and here’s how I solved them.

The first issue is not caused by CMake, but by pybind11. You can refer to this discussion: [flightmare issue #208](https://github.com/uzh-rpg/flightmare/issues/208). Simply follow the proposed solution or try running `pip install pybind11==2.12` to downgrade pybind11.

After that, you might encounter a second issue, as described in [flightmare issue #175](https://github.com/uzh-rpg/flightmare/issues/175). By applying the changes mentioned there, you should be able to resolve the problem completely.

Jinwei-Liu, 2024-10-16:
> > I encountered the same issues, and here’s how I solved them.
> > The first issue is not caused by CMake, but by pybind11. You can refer to this discussion: [flightmare issue #208](https://github.com/uzh-rpg/flightmare/issues/208). Simply follow the proposed solution or try running  to downgrade pybind11.`pip install pybind11==2.12`
> > After that, you might encounter a second issue, as described in [flightmare issue #175](https://github.com/uzh-rpg/flightmare/issues/175). By applying the changes mentioned there, you should be able to resolve the problem completely.
> > By the way, ` python_version == 3.6`
> 
> thank you for your record! what's the ubuntu version did you used? i encounter the same problem but can't make it

I used Ubuntu 20.04 to solve the issue.

zhaozilong2zl, 2024-10-16:
> I used Ubuntu 20.04 to solve the issue.

thank you for your quick reply! so you successfully install flightmare in ubuntu20.04 and python3.6 right?(congratulations！！！)

i am trying to do the same thing but fail for days

 did you complete the full installation merely by [Install with pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip) and the two changes(pybind==2.12 and master to main), i retry many times but keep failing T_T, looking forward to your reply, congratulates for your success again!

Jinwei-Liu, 2024-10-16:
> thank you for your quick reply! so you successfully install flightmare in ubuntu20.04 and python3.6 right?(congratulations！！！)
>
> i am trying to do the same thing but fail for days
>
> did you complete [...]

Yes, please proceed based on the documentation [https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip](url) and necessary adjustments, but just a gentle reminder: it should be **GIT_TAG "v2.12"**, not **GIT_TAG v2.12**, as mentioned in [this issue](https://github.com/uzh-rpg/flightmare/issues/208). Perhaps this is the reason why the installation didn't succeed for you.

Jinwei-Liu, 2024-10-16:
> > > > I encountered the same issues, and here’s how I solved them.
> > > > The first issue is not caused by CMake, but by pybind11. You can refer to this discussion: [flightmare issue #208](https://github.com/uzh-rpg/flightmare/issues/208). Simply follow the proposed solution or try running  to downgrade pybind11.
> > > > After that, you might encounter a second issue, as described in [flightmare issue #175](https://github.com/uzh-rpg/flightmare/issues/175). By applying the changes mentioned there, you should be able to resolve the problem completely.
> > > > By the way, `pip install pybind11==2.12`` python_version == 3.6`
> > > 
> > > 
> > > thank you for your record! what's the ubuntu version did you used? i encounter the same problem but can't make it
> > 
> > 
> > I used Ubuntu 20.04 to solve the issue.
> 
> thank you for your quick reply! so you successfully install flightmare in ubuntu20.04 and python3.6 right?(congratulations！！！)
> 
> i am trying to do the same thing but fail for days
> 
> did you complete the full installation merely by [Install with pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip) and the two changes(pybind==2.12 and master to main), i retry many times but keep failing T_T, looking forward to your reply, congratulates for your success again!

Yes, please proceed based on the documentation for [Install with pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip) and make the necessary adjustments: first, change the file `flightlib/cmake/pybind11_download.cmake` from `master` to `v2.12`; second, change `flightlib/cmake/gtest_download.cmake` from `master` to `main`. Maybe you can also post the error message.

Jinwei-Liu, 2024-10-16:
> thank you for your quick reply! so you successfully install flightmare in ubuntu20.04 and python3.6 right?(congratulations！！！)
>
> i am trying to do the same thing but fail for days
>
> did you complete [...]

Yes, please proceed based on the documentation for [Install with pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip) and make the necessary adjustments: first, change the file flightlib/cmake/pybind11_download.cmake from master to v2.12; second, change flightlib/cmake/gtest_download.cmake from master to main. Maybe you can also post the error message.

I tried again and confirmed that I only need to modify these two documents in ubuntu20.04 or ubuntu22.04 and python3.6 .

zhaozilong2zl, 2024-10-16:
> Yes, please proceed based on the documentation for [Install with pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip) and make the necessary adjustments: first, change the file [...]

thank you for your time for retrying the installation again!

however, i still got these errors below

1, after "pip install ." in /flightlib, got this failed to build opencv-python error
[image]
and i found an issue perfectly match my error infomation in [#209 ](https://github.com/uzh-rpg/flightmare/issues/209#issuecomment-2269606493), like [AnasAbdelR](https://github.com/AnasAbdelR) point that the issue may caused by unproper opencv-python version, so i try "pip install opencv-python==4.6" and "pip install opencv-python-headless==4.6" then, both install succesfully
[image]
and then i retry "pip install ." in flightlib, there is no error!
[image]

but when i "pip install -e ." in  flightrl (because pip install . will rise "No module named 'rpg_baselines.common'" error then), and then "python3 run_drone_control.py --train 0", i will get the dump error below (it confused me because i encounter this error before when i try to use python3.7 to install flightmare, at thar trail i got the exactly error)
[image]

so anyway, i still can't run the flightmare at last, maybe i should change another environment for my research haha T_T

soooooo much thanks for your time, i am great appreciate!!!! wish you can get your ideal experiment result!

Jinwei-Liu, 2024-10-16:
Yes, I also encountered the issue you mentioned. I originally thought you would solve it, so it's great to hear that you've managed to resolve the OpenCV-Python error. For your last issue, you can solve it by running:

`
pip uninstall ruamel.yaml
pip install ruamel.yaml==0.16.13
`

Finally, you will get it working. 

[image]

zhw970623, 2024-10-27:
> > Yes, I also encountered the issue you mentioned. I originally thought you would solve it, so it's great to hear that you've managed to resolve the OpenCV-Python error. For your last issue, you can solve it by running:
> > `pip uninstall ruamel.yaml` `pip install ruamel.yaml==0.16.13`
> > Finally, you will get it working.
> > ![image](https://private-user-images.githubusercontent.com/78484191/377035310-f5973365-f7c5-4b19-80f1-a3487ffd538b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkwODY3MzEsIm5iZiI6MTcyOTA4NjQzMSwicGF0aCI6Ii83ODQ4NDE5MS8zNzcwMzUzMTAtZjU5NzMzNjUtZjdjNS00YjE5LTgwZjEtYTM0ODdmZmQ1MzhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE2VDEzNDcxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzYzVlNWJjYTQ3MzI3Y2I3YjBlZDZiNzEyZmYxOGRiZjZjYzg5ODkyYTI1ZDJmYTEwNGY1ZDY2OGZmZTAzOGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.LqONr3XlmbGu4UJ38YGjogLP7oGYOOf2Ay6dFl1R5u0)
> 
> follow your command i successfully run the demo finally! can't be more appreciate for your kind instruction ! you save me from desperate!!!!!!!!!!!!!!!!!!!!!!!!!!
> 
> so for the afterward researchers who struggle to install flightmare in ubuntu20.04 and python3.6, please just follow our discussion and try one more time before you give up!
> 
> thank you Jinwei !!!

Have you ever encountered such an error when installing flightmare, I fell into the trap of pain just like you

 Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/anaconda/envs/flightmare/bin/python -u -c 'import io, os, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-p46yvcrt/setup.py'"'"';mp/pip-req-build-p46yvcrt/setup.py'"'"';f = getattr(tokenize, '"'"'open'"__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools imp)'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();e __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-z8anhpai
       cwd: /tmp/pip-req-build-p46yvcrt/
  Complete output (207 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 11.4.0
  -- The CXX compiler identification is GNU 11.4.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- Using system provided Eigen.
  /usr/local/include/eigen3
  -- Eigen3 include dir: /usr/local/include/eigen3
  -- Getting Pybind11...
  正克隆到 'pybind11-src'...
  fatal: 无法访问 'https://github.com/pybind/pybind11.git/'：Failed to coom port 443 after 133288 ms: 连接超时
  正克隆到 'pybind11-src'...
  切换到一个新分支 'v2.12'
  -- Pybind11 downloaded!
  -- pybind11 v2.12.1
  -- Found PythonInterp: /opt/anaconda/envs/flightmare/bin/python (found "3.6.13", minimum required is "3.6")
  -- Found PythonLibs: /opt/anaconda/envs/flightmare/lib/libpython3.6m.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  -- Found OpenCV: /usr (found version "4.5.4")
  -- Found OpenMP_C: -fopenmp (found version "4.5")
  -- Found OpenMP_CXX: -fopenmp (found version "4.5")
  -- Found OpenMP: TRUE (found version "4.5")
  INFOBuild time could be improved with ccache!
  INFO    sudo apt install ccache
  -- ======> Setup Compilation
  -- Enabling fast optimization flags!
  -- The activated CXX RELEASE configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -marc
  -- The activated CXX DEBUG configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -g -fopenmp
  -- ======>  Setup Build
  -- Getting gtests...
  正克隆到 'googletest-src'...
  已经位于 'main'
  -- gtests downloaded!
  -- Looking for pthread.h
  -- Looking for pthread.h - found
[... 132 lines cut ...]
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/dist.py",_commands
      self.run_command(cmd)
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/dist.py",_command
      cmd_obj.run()
    File "/opt/anaconda/envs/flightmare/lib/python3.6/site-packages/wheel line 299, in run
      self.run_command('build')
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/cmd.py", command
      self.distribution.run_command(command)
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/dist.py",_command
      cmd_obj.run()
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/command/b5, in run
      self.run_command(cmd_name)
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/cmd.py", command
      self.distribution.run_command(command)
    File "/opt/anaconda/envs/flightmare/lib/python3.6/distutils/dist.py",_command
      cmd_obj.run()
    File "/tmp/pip-req-build-p46yvcrt/setup.py", line 59, in run
      self.build_extension(ext)
    File "/tmp/pip-req-build-p46yvcrt/setup.py", line 91, in build_extens
      build_args, cwd=self.build_temp)
    File "/opt/anaconda/envs/flightmare/lib/python3.6/subprocess.py", linall
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--co '--', '-j4']' returned non-zero exit status 2.
  ----------------------------------------
  ERROR: Failed building wheel for flightgym

zhaozilong2zl, 2024-10-27:
> Have you ever encountered such an error when installing flightmare, I fell into the trap of pain just like you
sorry i didn't meet the same error.
the error info seems like a conflict of different version of zmqpp packages? ( in your message,'/usr/bin/ld: warning: libzmq.so.3, needed by /usr/lib/gcc/x86_64-linux-86_64-linux-gnu/libzmqpp.so, may conflict with libzmq.so.5')
maybe reinstall zmqpp could help