uzh-rpg/flightmare #213: Erro when run `roslaunch flightros camera.launch`

JzKosann, 2025-03-21:
when I ran `roslaunch flightros camera.launch`, I got below in my terminal. Wt can i do with this problem? It is the same situation when I added camera in my own code. The `[camera-3]process` will be killed
`(flightros_env) (base) jz@jz-ROG:~/Documents/ads_fpv_ws$ roslaunch flightros camera.launch 
... logging to /home/jz/.ros/log/7177140c-05ff-11f0-a28a-d5e1ddd9c5e4/roslaunch-jz-ROG-119041.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jz-ROG:41585/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    camera (flightros/camera)
    rpg_flightmare_render (flightrender/RPG_Flightmare.x86_64)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [119052]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7177140c-05ff-11f0-a28a-d5e1ddd9c5e4
process[rosout-1]: started with pid [119066]
started core service [/rosout]
process[rpg_flightmare_render-2]: started with pid [119069]
process[camera-3]: started with pid [119074]
process[rviz-4]: started with pid [119076]
[UnityBridge]  Initializing ZMQ connection!
[UnityBridge]  Initializing ZMQ connections done!
1 0 0
0 1 0
0 0 1
[UnityBridge]  Trying to Connect Unity.
[...........................................................................................................................................................................................................[UnityBridge]  Flightmare Unity is connected.
[camera-3] process has died [pid 119074, exit code -11, cmd /home/jz/Documents/ads_fpv_ws/devel/lib/flightros/camera __name:=camera __log:=/home/jz/.ros/log/7177140c-05ff-11f0-a28a-d5e1ddd9c5e4/camera-3.log].
log file: /home/jz/.ros/log/7177140c-05ff-11f0-a28a-d5e1ddd9c5e4/camera-3*.log`

Zhao-Zipeng, 2025-03-25:
I also encountered the same problem, have you solved it?

JzKosann, 2025-03-25:
> I also encountered the same problem, have you solved it?

Not yet, but when the first time I build the project, I successfully run the racing node with camera. But it failed after a month, I guess it may blame to some software updated automatically in Ubuntu 20.04.

JzKosann, 2025-03-28:
> I also encountered the same problem, have you solved it?

friend, I got the solution. There are two opencv with different versions in my computer (4.2 & 4.6). They confict with each other and kill the pid in this code `
```c++
          // Tell OpenCv that the input is RGB.
          if (cam.channels == 3) {
            cv::cvtColor(new_image, new_image, CV_RGB2BGR);
          }
```
src/flightmare/flightlib/src/bridges/unity_bridge.cpp: 263-266

When I rm cv4.6, I got the img.

[image]